cecio/USBvalve #4: Feature request: Port to Flipper Zero.

wanderling, 2023-07-16:
This would be a great tool to port to Flipper Zero if possible. I know Flipper Zero has the capability to become a USB host, not sure if it can be the right type of USB client though.

Thanks for USBValve! This is great how it is even if porting it is not possible!

cecio, 2023-07-16:
Thanks for the kind words!

Well, TBH Flipper Zero was the original choice when I started to think to build this tool and I started to look into the possibility to do this. But I faced some doubts and issues during this initial assessment:

- the SDK of the Flipper Zero is not straightforward to use like the Pi Pico one. Also is still under heavy development and quite often there are changes impacting portability of the apps (may be this is changed now)
- I have some doubts about the hardware: to run the USB host I use almost entirely one of the two cores of the Pi overclocked to 240Mhz (all the power is used for bitbanging).  The only core available on the Flipper is 32Mhz and the other one is dedicated to the firmware. Not sure if it can keep up the load
- I'd like to build something cheap. Flipper Zero is great (I have one and I find it very useful), but it is not cheap. 

That said, I'd like to keep this open, because I may give a second look to it, may be the doubts I had were not justified :-)

Thanks!

csdragon83, 2023-08-28:
If there's something to test in the future: I also have a Flipper Zero :-)

cecio, 2023-08-28:
Thanks!

Yes, I'm thinking to try out something for Flipperzero...so may be I'll ask your help :-)
Thanks!

cecio, 2024-03-22:
Had a deeper look into it, and technically speaking it should be entirely possible. Now it's just a matter of learning how to use the Flipper SDK and find some time ;-)

ianmaddox, 2024-03-23:
Another +1 for F0

markthwain1797, 2024-05-14:
I'd greatly appreachiate if this would become a thing.

cecio, 2024-05-14:
thanks for your interest @markthwain1797 @wanderling 

To be honest, I don't have too much time right now, so I don't think I'll be able to work on this in the short-mid term.