cecio/USBvalve #5: KillerUSB

cecio, 2023-07-18:
Chances are that it will be destroyed, so we can think it as a **sacrificial device** ;-) 
...it costs a few bucks :-)

Better this than your main PC!

**Need to be tested!**

csdragon83, 2023-08-28:
I also guess that it will be fried... 

For USB-Kill I recommend:
https://usbkill.com/collections/usbkill-accessories/products/usbkill-shield 
(even when the old one was cooler... you were able to do sparks with it)

cecio, 2023-08-28:
ah, I didn't know it existed! Nice!

Zorakie, 2024-03-22:
Hi, this my question is somewhat related.

If I add the USB-A for passthrough testing, and for some reason, that device will also sends power (like battery operated device, or something that is also plugged to the wall, say, USB-hub that also has power passthrough and is connected to laptop charger), is there any on-board protection for this power coming from the other side, or it will just fry the device?

Thanks! :)

cecio, 2024-03-22:
Usually host devices (even if battery operated) do not send power over the USB connection, they get power from it.
During my tests I used a lot of them and I never found one sending power over that line. 
This does not exclude that a malfunctioning one (or a malicious one) can do it. 

Looking at the Raspberry Pi Docs, I understand that the VBUS pin (the one attached to the power of the USB Host port) does not have a diode, so power coming in MAY damage the device (probably this depends on a lot of factors). A possible solution could be to attach the line to the VSYS pin, which actually have a diode. But in this case you may open an opposite issue: VBUS gets power from the USB Pico port, so it will be for sure at 5v and it will not damage your host. VSYS deliver the external power voltage (if you are powering the Pico externally) and that could damage your USB Host device (but only if you are powering it externally).

Side note: I see that you are referring to the USB-A as "passthrough" port, but it is not really that: the two USB ports are independent.

cecio, 2024-03-22:
Giving a deeper look at the Pico schematics, may be that even VSYS is not a solution in this case: if I'm not wrong the diode protects from backfeeding into the USB power source.
The real solution here is to apply an external diode probably.

frankrpeters, 2024-05-14:
As I also mentioned in the ATTiny/USBKill issue, a few TVS diodes and a polyfuse might protect the device and connected PC from damage.