cecio/USBvalve #8: 2 devices not recognized

csdragon83, 2023-08-21:
Hi!

First of all: you did a great job creating this little device...
I meanwhile made 3 if them...

But 2 hacking devices from me were not recognized...

1.:
"Duckhunter" is not recognized...

Duckhunter is a app within Nethunter that interprets Duckyscripts.
Nethunter is a Kali Linux arsenal that can be installed to Android devices..

In Nethunter there is the function "USB arsenal". With it you can enable the function "HID". 
When I now start an attack at a computer the app "Duckhunter" interprets the Duckyscripts and types them in via the connected USB cable.
For that the android setting "USB Debugging" needs to be activated on the Android device.

Now I thought that the USB valve would also recognize the key strokes...
But it completely doesn't...
It doesn't recognize when I change my phone to HID mode and it doesn't recognize the key strokes...

2.:
The second device is a little "CJMCU" - a little Mini Atmega32U4 development board for about 7€ that also can do keystrokes... 
At the moment I have set it to type in 0000 to 9999 but the USB valve also doesn't recognize the keystrokes of this little device...

Maybe you could add a function so that this devices also  can be recognized...

Best regards
Csdragon

cecio, 2023-08-21:
Thanks for the kind words, and for opening the issue.

A couple of questions:

- are the devices not recognized at all, or just the keystrokes? I mean, when you plug the device, do you see an `HID Device` message on the OLED screen, or nothing at all?
- for the NetHunter app, did you tried to switch before to HID and then (after that) plug the device into USBvalve?
- the CJMCU device is also named "CJMCU-Beetle"? I can try to buy one to do a couple of tests

Thanks,

cecio, 2023-08-21:
Ok, I think I found something, not sure if this is the same issue you are experiencing, but may be:

it looks like the overclocking of the CPU is in some way decreasing the compatibility with some HID devices, not sure why. May I send you a debug version of the firmware with a 120Mhz firmware (let me know if 32 o 64 lines screen) and may be you can do some testing with your devices?

Thanks,

csdragon83, 2023-08-22:
> it looks like the overclocking of the CPU is in some way decreasing the compatibility with some HID devices, not sure why. May I send you a debug version of the firmware with a 120Mhz firmware (let me know if 32 o 64 lines screen) and may be you can do some testing with your devices?

Would be no problem!
I've got a 32 lines display.

csdragon83, 2023-08-22:
To the CJMCU device: I hope I can manage to have a look at it this evening.

cecio, 2023-08-22:
No worries at all, take your time. And thanks for your help!

In the meantime, here is the link for a new test version (0.13.0): https://icedrive.net/s/8SbF9bD7N27k1vGYBtDVya9FCjzk

Let me know how it works with the two devices.

Thanks again.

csdragon83, 2023-08-28:
Hi!

Installed it to my USB-Valve...
Well... most of the time it is recognized now... but about one time in 20-30 times restarting it an then plugging it in it is not recognized...

I can plug it in, unplug it a lot of time - nothing on the display...
Reboot by button of the USB-Valve and it is recognized...

To the CJMCU device: sorry have a big construction site at home at the moment... was not able to have a look yet...

csdragon83, 2023-08-30:
The Hardware I use is a Digispark hw-018
Costs about 2,50€ at ebay.

cecio, 2023-09-02:
I ordered one, should receive it next week. I'm curious to check it out.

Thanks,

cecio, 2023-09-05:
I received the device and to be honest I'm a bit puzzled:

the device itself has something wrong: I tested it in two different laptops (a Dell with Linux and a Lenovo with Windows) and none of them recognize it. Even if it receives power (and the led turns on), the device is not recognized if inserted in the built-in USB ports of both the systems. 
I think the PCB is too thin and it does not fit correctly in the USB ports. If I compare it with `Adafruit Neokey` (https://www.adafruit.com/product/5020), it is almost 0.5mm less, which is a lot. 
`Adafruit Neokey` works perfectly with all my laptops and USBvalve as well.

I had to use a specific USB extension (but a couple I had were not working as well) and with this I have the Digispark device recognized 50% of the time by the laptops and by USBvalve as well. But it is not reliable in any case.

May be I received a defective device (but the PCB thickness is an issue in any case)...otherwise I don't think this is a USBvalve issue. It looks like a connection issue.

cecio, 2023-09-05:
May I ask you a favor?

I have a new version of the firmware: https://icedrive.net/s/NT52Z44k2k4PbD2YxvbtYSyiNyVY

Could you please test it with NetHunter? I'd like to check if this new version keeps the compatibility.
You can also check it with the Digispark device, even if I don't expect it works with that.

Thanks a lot!

twhp-electronic, 2023-10-18:
Sorry... Just saw today, that you updated to a new version.
Works very well with NetHunter.
As soon as I change USB to HID it is recognized.

cecio, 2023-10-18:
oh no worries at all... :-)
Thanks for the checks!!

Regarding the Digispark, please try, but I'm pretty sure it will not work. The reason why this is not recognized is still puzzling me...or may be I have a defective device.

Thanks a lot!

csdragon83, 2025-05-02:
I found a new device that is not recognized... 

https://github.com/joelsernamoreno/EvilCrow-Cable

cecio, 2025-05-02:
Thanks for letting me know. I ordered one, as soon as I'll get it, I'll have a look.

csdragon83, 2025-05-06:
If you need example code: please let me know!

cecio, 2026-01-03:
Not yet unfortunately, but I hope to have some time in the next days to look into it.