cecio/USBvalve #14: pcb board ordering suggestions

tbyoda, 2024-03-20:
hello great project and seen this on LTT and want  to build this great project. However i have never ordered hobbyists pcbs. whom are you using for this project? i wouldn’t mind ordering a few boards if others want to go in or if you have some boards left over buy from you. please let me know. i have ordered pico and screen.

ondrejhonus, 2024-03-20:
Hey man, I just ordered 5 pcbs yesterday for total of 3$ incl. shipping from [jlcpcb.com](https://jlcpcb.com/?from=VGS&gad_source=1&gclid=CjwKCAjw7-SvBhB6EiwAwYdCAeGW0AIzZreAneSbFapvBUYa0NGw-5qVMokNbCpnxBy_ON4QMvU2IBoC1_EQAvD_BwE). You just download the PCB zip from this github dir and you upload the zip to the jlcpcb website. That’s basically all, it’s easy.

tbyoda, 2024-03-20:
Thank you very much , I thought it would be easy but wasn’t totally sure. I
can’t wait to make this project will be fun learning process.
Robert

timothy-fyi, 2024-03-26:
> Hey man, I just ordered 5 pcbs yesterday for total of 3$ incl. shipping from [...]

you didn't need to get it assembled/attach a BOM? that's my current problem. I know how up upload the Gerber files to jlcpcb, but I wasn't sure if I needed it assembled as well

tbyoda, 2024-03-26:
No i don’t need any assembly. I am a RF Engineer by trade so this will be
easy.

Sin-Thor, 2024-04-18:
> No i don’t need any assembly. I am a RF Engineer by trade so this will be easy.
> […](#)

I've never done anything like this before or ordered a pcb before, so not even sure if the pads to solder too are built into the board or not. If I want this usb pcb to be ready to go, out of the box, ready to be assembled with the raspberry, screen,  and usb port, which Delivery Option do I select between 'Single PCB', 'Panel by Customer' and 'Panel by JLCPCB' ?