cecio/USBvalve #17: Possible with a USB-C version?

nmcbride, 2024-03-21:
I would like to build one of these but with USB-C on both sides. 

I was thinking a Pro Pico and a USB-C on the other side. 

Would this be possible?

cecio, 2024-03-21:
Hey, thanks for reaching out.

I had a look around for Pico USB-C versions: there are some of them and I noticed that they have a slightly different form factor (especially for the debug PINs), so probably you can not use any of the PCB nor the spacer for the OLED screen.

But if you are going to build your own version with a breadboard, it should work without problem. Remember that USB-C (for the host port) requires a resistor, so you should add it.
Actually I tested it some months ago (with a USB-C breakout), as you can see from the picture here: https://twitter.com/red5heep/status/1640812839525928986

I thought to create a PCB design with USB-C as Host port, but there is something that makes me doubtful about it: since the connection requires a resistor, I should do it with a SMD component and this is going to make the build of the board more difficult for beginners, so I never did it. But I might change my mind about this.

Zorakie, 2024-03-21:
Hi, I have just got my parts delivered and I genuinely thought it has USB-C as the main port (really, who uses micro these days on smart devices, where this matters). That being said, have you please tested, if I buy a simple Micro USB (male) to USB-C (female) reduction, and I plug it to PC, shopping mall charger, whatever, that it will work as expected? Thanks!

cecio, 2024-03-21:
hey, @Zorakie. I don't know why but the official Raspberry Pi Pico is still delivered with micro USB.
And, yes, with a converter it will work as expected!

kub3let, 2024-03-22:
There are third party boards like this https://de.aliexpress.com/item/1005006055009344.html with USB C.

Personally I would prefer to have a PCB version that has both USB A like it is now and USB C on the back so you can use either.

I Just ordered PCB's and will test if it works with the Tenstar Board.

cecio, 2024-03-22:
Thanks, let me know how it goes: the Tenstar Board seems to have a slightly different form factor.

I saw that some other "unofficial" releases have the same form factor with USB-C, like
https://www.aliexpress.us/item/3256805935885677.html?spm=a2g0o.productlist.main.17.2f27PZpZPZpZ96&algo_pvid=fd68471b-abf6-42ca-8f80-9afd0b7c5cdb&algo_exp_id=fd68471b-abf6-42ca-8f80-9afd0b7c5cdb-8&pdp_npi=4%40dis%21USD%212.63%210.99%21%21%212.63%210.99%21%402103894417111047216746036ecc03%2112000035853862386%21sea%21US%210%21AB&curPageLogUid=JwbUbPeVELpE&utparam-url=scene%3Asearch%7Cquery_from%3A

kub3let, 2024-04-03:
PCB's arrived today, but the USB port is not working, after double checking and checking with a multimeter I noticed there isn't any ground trace for the USB port on the PCB ??

I will hotwire the ground, but it should be added to the gerber files.

Since this pico board comes with a RGB LED you don't even need the display technically or you can use them both, in that case the led works as background light for the display which is pretty cool.

[image]

entertheFinn, 2024-04-03:
This is what I am currently using as my USB C option. Unfortunately still working on getting the background to work like it does on the 1.28 inch screen. This is the o.99 inch B edition of the Waveshare RP2040. It accomplishes my USB C needs and is thinner than a lighter so you know I'll have multiple of these all over the place. That suitcase gets one. That toiletry bag. EDC gear. I have not tested inspecting traffic/payloads tho.[image]

nmcbride, 2024-04-03:
I was thinking of the pro pico as that is usb-c by default but I like the dual option with both the usb-c and usb-a.