cecio/USBvalve #18: 3D print enclosure?

Zorakie, 2024-03-21:
Hi! Any chance anyone has any blueprints for some enclosure? I have bought Pi Pico and 128*32 display and now I was thinking about how to put it together. I have ordered also some small plastic enclosure (Zonepi case, but it seems local) that fits more or less exactly the Pi itself.

Now, because the Pi has MicroUSB input, and in the reality, you will be testing rather USB-Cs (e.g. plug it to a mall phone charger, etc.), you will need a reduction. Then, there is this thing with adding the USB-A for passthrough testing, and then there is this thing with the OLED display.

I was already thinking around, how to put it together and it seems like it will be quite janky.

What I am looking for:
Enclosure that will fit the Pi itself, will have hole for micro, will have hole for USB-A, will have proper place for display, that should be aligned with the box surface, and it should have some clips/compartment that will hold micro(m)->C(f) and A(m)->C(f) if you will need to switch between interfaces on-the-go.

Unfortunately, I do not have 3D printer, nor any experience with 3D modelling ...

So, anyone any ideas?

cecio, 2024-03-21:
Unfortunately I have very limited modelling capabilities as well, so the only thing I can do here is to forward the request to the community: if anyone want to contribute with something like this, it will be more than welcome :-)

Zorakie, 2024-03-21:
That's why I am asking here, but if you would be willing to ask, on twitter, or other social media of yours, and then include the blueprints to this repo somehow, considering somebody will make anything, that would be totally amazing! Thank you!

gamecioep, 2024-04-10:
I recently put a USBvalve together myself and am currently creating/printing an enclosure of it.
I can share this if you want (once I know that it fits correctly), but don't expect to much of it.
It is a very simple design (nothing too fancy) nor do I think that my design will fully accommodate every feature you want of it (specifically being able to place the cables inside.).
My current desing assumes that you used the PCB version.

SinStalker, 2024-04-19:
I found this that I plan to print once I put everything together and make sure it works. Maybe this will work for you.
[](https://www.thingiverse.com/thing:6557010/comments
)

cecio, 2024-04-19:
See Issue #36 a very nice enclosure has been added.

closed 2024-04-20 by Zorakie