cecio/USBvalve #19: RP2040 GEEKs Development Board

KonTy, 2024-03-23:
@cecio do you think RP2040 GEEKs Development Board can be modded to work with your firmware, as oppose to building a custom board? It seems that it would be much less painful if we could just mod this board and make it work?

https://www.ebay.com/itm/315215910989?mkcid=16&mkevt=1&mkrid=711-127632-2357-0&ssspo=cdxlfgc5ryy&sssrc=2047675&ssuid=IFHFWV1lQai&widget_ver=artemis&media=COPYM1j

cecio, 2024-03-23:
Hey @KonTy , thanks for you interest in the project!

If you are interested in the USB mass device functionalities, the only issue to solve here is the LCD compatibility: I expect it will not work out of the box, but we should find a compatible library to use for that LCD (may be the one used for the PI Pico Watch is ok, but I expect that the PINs used will differ).

If you want to have also the USB HID detector, it's a bit more difficult: remapping the PIN used for D+ and D- is trivial, but I don't see any external PIN for VBUS or VSYS (which should be 5v). The only power PIN I see is 3.3v. For sure the PIN is somewhere on the board, but you should bring it out in some ways, so it would be a bit cumbersome to do.

KonTy, 2024-03-23:
@cecio I just bough that board, I will start looking into porting the fw to it. I think you are right I will try to get basic functionality first and then we could discuss USB HID functionality. TBH I think itwould give the community a huge boost if we could build this on top of a packaged solution like the one above so you won't have to build, solder things.

cecio, 2024-03-23:
Keep me posted on that! Thanks