cecio/USBvalve #20: [Enhancement] device detected message

Zorakie, 2024-03-24:
When I first plugged my USBValve in, I thought my USB-A didn't work (bad solder, switched pins - as I was soldering it upside-down) or busted board because of my soldering ...

Well, if I put in for example a keyboard, it correctly starts detecting HID, but if I plug in a simple USB drive, there is nothing that would indicate that the device is actually plugged in, detected, or the board just knows about it in any other form (even if it would be just a detection that it consumes power on PIN 40.

It would be nice that if whenever a device is plugged in, there would be some info message first, before any error (if any) states would pop up.

cecio, 2024-03-24:
I'll look into that. Thanks!

cecio, 2024-03-25:
I added an initial support for this with release `0.16.0`.

closed 2024-03-25 by cecio

Zorakie, 2024-03-25:
Works like charm. Thanks a lot!!

cecio, 2024-03-25:
Thanks to you for the suggestion!