cecio/USBvalve #21: Ampel anzeige Ws2812b LED

TryBreakFixAgain, 2024-03-25:
Would it be possible to integrate a single Neopixel or Ws2812b control to make it easier to read for older or less tech. people?

I am thinking of:
Off = nothing connected
Blue = Something connected
Green = everything ok
Yellow = Attention
Red = something bad found

cecio, 2024-03-25:
Hey, thanks for your input.

Your idea is for sure interesting, and not to difficult to implement from a SW point of view. May be for HW is a bit more difficult: adding a Neopixel is going to complicate the DIY build.
I'm wondering to keep your idea for the pre-built board.
Anyway, thanks again for the idea. I'll try to integrate this somehow.

TryBreakFixAgain, 2024-03-25:
the hardware side should be less of a problem, 5v, gnd and one gpio. Single LED are now quite easy and cheap to get, I've tinkered with them a bit, but the system freezes when I try to control the led, but I'm a complete beginner when it comes to programming and not skilled enough to debug it

Great project

TryBreakFixAgain, 2024-03-28:
I tinkered a bit and found a quick and dirty solution based on NeoPixelConnect.

[https://github.com/MrYsLab/NeoPixelConnect](url)

My code is totally experimental and not professional, I don't know if it affects other functions. according to my tests everything seems to work so far.
I would be pleased if you could check whether my code affects the function. Or you might even see a possibility to include an improved version of my code in your project.  I would provide a modified PCB and a case design if my code does not affect the functions.

I have used a single WS2812B.
PIN36 (3V3) of Pi -> VCC
PIN38 (GND) of Pi -> GND
PIN14 of Pi -> Din
Yes I have only tapped the 3V3 instead of the 5V but it is enough to power the LED.

In order not to contaminate the whole code, I only modified the printout(), which is not so nice, but quick and dirty.  

**Initialization**

    //START TryBreakFixAgain TrafficLight initialization
    #include <NeoPixelConnect.h>
    #define PixelPin        10 // PixelData Pin
    #define PixelNum 1 // Pixel count
    NeoPixelConnect pixels(PixelPin, PixelNum, pio0, 1);
    //END TryBreakFixAgain TrafficLight initialization

**modified printout()**

    void printout(const char *str)
    {
      display.print(str);
      // START TryBreakFixAgain TrafficLight
      String tlString = String(str);
      tlString.replace(String(char(10)), "");
      tlString.trim();
      if (tlString.substring(0,3) == "[+] ") {
        pixels.neoPixelFill(255, 255, 0, true);
      }
      else if (tlString.substring(0,3) == "[++") {
        pixels.neoPixelFill(0, 255, 0, true);
      }
      else if (tlString.substring(0,3) == "[!]") {
        pixels.neoPixelFill(255, 153, 51, true);
      }
      else if (tlString.substring(0,3) == "[!!") {
        pixels.neoPixelFill(255, 0, 0, true);
      }
      else {
        pixels.neoPixelFill(0, 0, 0, true);
        pixels.neoPixelClear(true);
      }
      // END TryBreakFixAgain TrafficLight 
    }

cecio, 2024-03-28:
Thanks for your proposal.

I think your code is good, but it can be simpler in our case. It should work fine (I didn't tried it), but instead of using strings function to check the printout buffer it should be enough to check the booleans used to report anomalies:

```
boolean readme = false;
boolean autorun = false;
boolean written = false;
boolean deleted = false;
boolean written_reported = false;
boolean deleted_reported = false;
boolean hid_sent = false;
boolean hid_reported = false;
```
Also, the logic you are using is a bit strange: you are setting yellow for `[+]` messages, but that mean that yellow will be always set, since the `selftest` is printed with `[+]`. I would review it a bit.

I'm not sure I'm going to add this to the main branch, since I'd like to keep the HW build as simple as possible (and as I said I'll add something with the on-board led), but if you are going to work on it and create PCB and other things, I can create a dedicated branch for it, if you wish.

TryBreakFixAgain, 2024-03-29:
thank you very much, i am new to coding, i belong more to the 3d printing and soldering section.
i think it's very cool that you want to keep everything as simple as possible.

i'll keep tinkering with the idea for myself, it's a good opportunity to learn.

if it ever works exactly the way i want it to, i will let you know and of course make it available to everyone.

Thank you very much for your help and suggestions.

cecio, 2024-03-29:
Thanks to you!! 
And if you have any question, just get in touch.

TryBreakFixAgain, 2024-04-11:
I have taken the liberty to customize the code a bit to suit my needs,

I made all display output quickly configurable with json as I see the use of other languages in my environment.

On top of that I have built a traffic light system.
I am using a Ws2812b DOT (https://www.amazon.com/dp/B088K8DVMQ) on pin 10 (GPIO7).

All changes are documented in the code and can be found via TryBreakFixAgain and tbfa_.

I use ArduinoJson 7.0.4 from Benoit Blanchon
I decided to use the lib and json because I have future changes in mind that I want to make.
Since I am not a programmer, I declare the code as experimental.

 Of course I make the idea and my changes available for adoption into the main code and for general use.

[USBvalveLanguageLightMod.txt](https://github.com/cecio/USBvalve/files/14948277/USBvalveLanguageLightMod.txt)

cecio, 2024-04-11:
Thanks a lot for sharing your work!

As I anticipated, I'd like to keep the build as simple as possible, so I'm happy to keep your mods here if someone wants to use them, but I'd like just to make a super simple mods on the main code to use the on-board led.

Thanks again for sharing this!