cecio/USBvalve #22: Usb A male

ondrejhonus, 2024-03-25:
Hello, I am thinking about building this project and I'm wondering if i can buy a male to male adapter (as shown in the picture) or perhaps solder a male usb A instead of the female one.

Sorry if this is a dumb question, but i think it would be better this way.

[image]

cecio, 2024-03-25:
Thanks for reaching out.

You can use any type of USB, but I'm not sure the male format is the proper one in this case: the `USB Host` port (the Female A one) is used to detect USB peripherals which may act as `HID` (keyboards, mouse, etc). Almost all these devices have male adapter, so you will have some difficulties to connect them.

I think it make more sense to think about a male adapter for the other port, used to connect and act as `Mass Storage` device, in order to detect malicious access to the dongle. But this is more complicated to do because the port  (micro USB) is the one provided by the Raspberry Pi Pico, so may be it's easier to use an adapter in this case.

Remember also that the two ports are not interchangeable: they have specific functions.

Let me know if something is not clear, or if you have other doubts

ondrejhonus, 2024-03-25:
By an adapter you mean male **usb-a to male usb-a** or **micro usb to usb-a**? Or perhaps some other one?

ondrejhonus, 2024-03-25:
Okay, I see now. I thought i could just power the pico over the USB-A port and plug it into a pc like that, I get it now, thank's for your help.

ondrejhonus, 2024-03-25:
I found what i need now :smile: 

https://www.amazon.com/UCEC-USB-2-0-Adapter-Male/dp/B00TAM0MZW
[image]

closed 2024-03-25 by ondrejhonus

cecio, 2024-03-25:
exactly! :-)