cecio/USBvalve #25: ATTiny Attacks / USBkill Attacks

NagelTuev, 2024-03-26:
I have played through a few attack scenarios and am absolutely delighted with the USBvalve.
However, USBvalve did not recognize a Rubber Ducky based on an ATTiny85 in the form of a Digispark Rev.3. 

Unlike other Rubber Duckies I have tried, windows does not identify the Digispark as HID but as a USB input device.
USBvalve does not respond at all.

Since Attiny's are very small and very cheap, I see more potential for attack in them than in other Rubber Duckies, simple instructions can also be found very quickly.

Because USBvalve is so cheap to produce, it is the optimal tool that everyone should have.
However, I still see a problem in USBkill attacks,
I think most users will power USBvalve from a computer to check USB sticks.
So the overvoltage from the USBkill would probably still cause damage to the end device. 
I'm not an expert on the subject, but I think galvanic isolation would certainly make the project too expensive and complicated.
Possibly a battery could avoid the problem, but I don't know how it reacts to the overvoltage.

I think a note "Supply USBvalve with a powerbank when testing sticks" should be the easiest way.

USBvalve is a fantastic project, thanks for the effort and sharing.

cecio, 2024-03-26:
First of all thanks a lot for your kind words, really appreciated :-)

ATTiny85: I'm aware of this, I bought one of this actually. I don't know if I have a defective device or not, but I barely have it working even on regular USB hardware: most of the times it drops error, it is not recognized at all. I try to flash it, it works one time than it stops again. So, before try to fix it on USBvalve I need to see it working regularly on real USB hardware.
TBH I didn't gave to this high priority so far, but I'll look into that.

Regarding USBKill: I'm not sure what could happen to the upstream device (battery or PC) if you attach a USBkill to the Host port. Probably the "safest" procedure (but I'm not recommending trying it), if you suspect a device is a USBkill, is to attach it to a completely disconnected USBvalve: I assume you will see a flash and some smoke coming out, and then the device will be completely dead. This could be your "canary" :-)

Thanks again.

Himitsu-NL, 2024-04-05:
Hi, not sure if this is helpfull, i bought a couple of ATTiny85 Digisparks ( 4 ) specifically to test the USBvalve, all of them work on the 4 systems i had on hand but are not seen by the USBvalve, the green led's on the Digispark lights up so it does receive power from the USBvalve.
I thought i had made an error in my hardware build before finding this thread.

This is an awesome project, keep up the good work :)

cecio, 2024-04-05:
Thanks for letting me know and thanks for your kind words!

cecio, 2024-04-10:
nice catch @NagelTuev  !
Thanks for letting me know this, I'll try to have a look.

Tz1rf, 2024-04-24:
Has anyone else tested using a power bank or similar device? I had a similar idea about using a power bank instead of a computer as it makes it highly portable and easier to use. So I decided to test this out 

If you use a USBValve with a power bank, portable battery or similar device, the USBValve powers on and a couple seconds after I plug a USB Flash drive into the USB A port the screen shows mass storage, then the USBValve turns off. It doesn't appear to cause any damage to the USBValve or the power bank as I am able to repeat the process many times and both are still working. I am curious if others will get the same result or if its something specific to the power bank I am using.

cecio, 2024-04-24:
Some power banks have an auto shout down if the power drained is not enough. Since USBvalve is not requiring a lot of power, may be that's why you see the device going off.

Tz1rf, 2024-04-24:
@cecio 

I hadn't thought about that. Let me try another unit and see if I get the same results. I assumed it would work since I use it to power a Raspberry Pi Zero and it will stay powered on until the battery depletes.

frankrpeters, 2024-05-14:
About the USB killers: maybe a few TVS diodes and/or a poly fuse might mitigate that problem? Would at least protect the USB valve itself as well as the connected computer. Would of course be still better if there was a way to detect this.