cecio/USBvalve #27: Buzzer Support?

LiPolymer, 2024-04-04:
Could you please add an optional feature that can notify the user with a buzzer when some suspicious behaviors _(e.g. HID device attached)_ have been detected.
_I'm not an Arduino developer, and I'm sorry I can't help with this amazing project._
(By the way, I designed a dedicated PCB for this project. Can I create a PR when I have tested it?)

cecio, 2024-04-04:
Hey, thanks for your interest into the project  and your feedback :-)

I think the buzzer could be a good idea, but as for another request about an RGB led, I'm a bit doubtful: I'd like to keep the design and build of the main board as simple as possible. Buzzer and additional LED can be useful, but are going to complicate the build process and even if they are not a big issue for someone experienced in building electronics, they can be difficult to manage for novices. So, for the time being, I'll keep this on-hold.

And yes, please if you have a dedicated PCB, open a PR. Thanks!

LiPolymer, 2024-04-05:
Thanks! :)
My PCB is in production.
I'll open a PR if it works well. :D
 
And about the buzzer, maybe you can make it an **optional** feature. Users can choose whether to weld and use it.  ;)

You can control its pitch by changing the PWM signal's frequency. 

I guess that a humble buzzer can make a big difference. :D

Have a nice day ;D