cecio/USBvalve #29: need some pointers please

copywizard, 2024-04-06:
Ok so i got my RP2040 watch from ali everything worked fine out of the box/bag.

But after trying to flash the usbvalve image to it or any uf2 image for that matter the usb device popped up and i dragged and dropped the uf2 on the watch and nothing happened!

So i tried numerous other uf2 files to no avail. now i tried using Arduino IDE to upload the raw data as a sketch and after hours of work i got it to finally upload to the watch i was so happy!

until i saw a black screen and nothing happening on the watch itself i unplugged it and plugged it back in now it identifies itself as a usb drive named MyDrive with two files on it AUTORUN.INF and README.TXT "that says ...nuke the entire site from orbit. It's the only way to be sure." 

I really need some pointers if possible? i would love to use the watch as an USBValve.
But why cant i for the love of god not get the UF2 file to flash the device?

I have tried it with a regular pico and rp2040 with all kinds of uf2 images and that works fine!
A few days ago i used this very same setup to flash a PICO with Jrunner to flash my xbox 360 which went very well.
But somehow this device won't work properly.

cecio, 2024-04-07:
Hey.

If you managed to get the "Mydrive" mounted, this means that the firmware has been uploaded correctly. Also the file you see are expected and are part of the filesystem itself.
If, despite this, you don't see anything on the screen, there are two possibilities:
- defective device (unlikely, did you see it working with the original firmware? Or can you upload the original firmware to test?)
- the device is not fully compatible with the one I used to create the firmware

Unfortunately there are a lot of different way to wire up LCD to the Pi Pico and if the one you bought is not the same as the one I used to create the firmware, it requires some modification to work. 

The best thing here is to obtain the schematics of the connection of your PI watch if available, so we can check the PINs used to connect the LCD and see why it is not working.

copywizard, 2024-04-07:
> Hey.
>
> If you managed to get the "Mydrive" mounted, this means that the firmware has been uploaded correctly. Also the file you see are expected and are part of the filesystem itself. If, despite [...]

Thanks for the quick reply i didn't expect there would be so much different possibilities for wiring a lcd to a pi pico.

I have added a picture of the labeled pins maybe this could help?

[image]

I also do not think its a defective device because i worked when i got it in but it would just not accept any UF2 files so i could not flash anything.

today i will try and to upload the original firmware if i can find it somewhere and hopefully i can use arduino ide to upload it i have no idea how to decompile a UF2 to its raw format or if this is even possible?

cecio, 2024-04-07:
Model looks the same as mine.

If you uploaded the firmware through Arduino IDE, you probably re-compiled it. Did you un-commented the following

```
// Uncomment the following to compile for the RP2040 based TFT round display
// https://www.raspberrypi.com/news/how-to-build-your-own-raspberry-pi-watch/
//#define PIWATCH
```

Otherwise you are uploading the wrong version. Anyway, the easiest way is to upload the UF2.

copywizard, 2024-04-07:
> Model looks the same as mine.
>
> If you uploaded the firmware through Arduino IDE, you probably re-compiled it. Did you un-commented the following
>
> ```
> // Uncomment the following to compile for the [...]

Thats it i think i just don,t know what i should comment and uncomment?

cecio, 2024-04-07:
```
//#define PIWATCH
```
should become
```
#define PIWATCH
```

Then compile and upload again

copywizard, 2024-04-07:
Nearly there i hope

i have this in there now

#define PIWATCH

#include <Arduino_GFX_Library.h>
#include "background.h"

#else

#include "SSD1306AsciiWire.h"

#endif

and i get this error

exit status 1

Compilation error: #else without #if

cecio, 2024-04-07:
No, you messed up something, it looks like you removed a line.

If you just un-comment that line, you should not get any errror. Start from the clean source and just un-comment line 24.

Then recompile (assuming you are using Arduino IDE with the proper libraries listed in the repo)

copywizard, 2024-04-07:
Copied the code from here 

https://github.com/cecio/USBvalve/blob/main/USBvalve/USBvalve.ino

And pasted it in Arduino IDE uncommented the piwatch part 

still the same error.

could it be that you say to uncomment "//#define PIWATCH"  but in your own code it says "#if defined(PIWATCH)"

cecio, 2024-04-07:
you are referring to line 31

Look at line 24. Commens in C are not `#` but `//`

So your line 24 should be like this to work:

```
#define PIWATCH
```

copywizard, 2024-04-07:
Damn that did the trick! thanks very much and sorry i think i just can't count :(

Still does not fix the flashing issue but that a whole other problem i have a screen right now!

cecio, 2024-04-07:
Great! Happy it worked! :-)
Thanks!

closed 2024-04-07 by cecio