cecio/USBvalve #31: uf2 file not flashing

Twanaki, 2024-04-09:
Hi,
I'm new to this, so I am probably doing something wrong but I don't know what.
I soldered an I2C OLED screen 128x64 and an USB A port onto a Raspberry pico.
I flashed the blink.uf2 file (from raspberry website) for testing purpose, it works. I simply put file into the mass storage that appeared.
Then I restarted with the BOOTSEL button pressed, and released when the mass storage appeared.
But when I try to flash the USBvalve-0.16.0-64.uf2 file, nothing happen, the folder stays open.
Any help would be much appreciated !

Twanaki, 2024-04-09:
I struggled a lot, but finally managed to get it work by installing Arduino IDE and everything necessary, and then flashing from there.

cecio, 2024-04-09:
Strange. I just tried to do the same, but it worked fine on my system.
I'm not sure what was causing the issue, but happy you managed to find out the solution.
Thanks for the update

Twanaki, 2024-04-10:
I don't know why it wasn't flashing on mine. I bought the Pi Pico on Amazon from a random seller.
Thanks for your tool, it's fantastic !

Zorakie, 2024-04-10:
Maybe it is lost in translation, but from what you wrote, I have an impression that you may be putting the file to a wrong storage.

There are 2 storages:
1) when you have this project running, it shows a storage with some content, but that is basically fake.
2) The storage where you need to upload the binary file is the one with BOOTSEL.

Maybe that behaves differently the first time ever the pico is started, but generally, you should:
1) Unplug pico
2) Hold BOOTSEL while plugging the microusb cable to your computer. Don't release the button until the pico is plugged in and the storage shows
3) storage should pop up (pico is now in the setup mode)
4) upload file there, the pico should automatically restart
5) unplug it and plug it again
6) TADAA, you should be running USBValve

DFranzen, 2024-04-13:
I am having the same issue. Tried via the usb storage and also via picotool, but no luck so far.
The example blink.uf2 flashes successfully, but with usbvalve the usb storage never unmounts and the raspberry does not reboot. Upon manual reset after several minutes of inactivity, it just goes back to the previously uploaded program.
I am using this unofficial rp2040 board, which has a usb C port and also an on board RGB LED, so would he pretty nice to have usbvalve running on it: https://www.aliexpress.com/item/1005006051120831.html

DFranzen, 2024-04-13:
Quick update, I thought to check a previouse version: version 0.16 flashes successfully in the normal way.

Zorakie, 2024-04-13:
Maybe there is something wrong with the unofficial board... jumping here just to confirm back that for official Pi Pico board, everything seems to be working. I have just flashed 0.17.0 via BOOTSEL/directory with no issues 🤷‍♂️

cecio, 2024-04-13:
Thanks for the test @Zorakie 

@DFranzen it sounds really strange. The presence of RGB LED on the unofficial board open to some possibilities, like incompatibility of the used PINs etc, but despite this, you should be able to flash: then may be the board does not work, but at least it should load the firmware and reboot.

DFranzen, 2024-04-23:
Another quick update, might be able to help others with this issue:
The uf2 file that is available in the Releases-Section on Github (V0.17) does flash and runs without any problems. The one I tried before is the one in the folder structure in the folder "firmware" (as described in the README). That one still does not flash. Indeed they have two different hashes:

$ shasum USBvalve-0.17.0-32_*
1760708087fd7cf4a5e6bac86bc7baa35600f0cd  USBvalve-0.17.0-32_release.uf2
6869c877579ffc58305fbc49b9a251c3a3b109ff  USBvalve-0.17.0-32_tree.uf2

Is that difference intended?

cecio, 2024-04-23:
That's weird: tbh I tried to replicate this, but now way:
I tried to download the file from the Releases, clone the repository, download it from the firmware folder, and I get always the same hash `1760708087fd7...`

Could you please check size and content of the other one? I don't understand from where is coming.

cecio, 2024-04-23:
are you sure that you downloaded it in the proper way? For example, going with the browser in the `firmware` folder and right-click on the file name and doing a "Save Link" is going to save an HTML doc, not the firmware file.
Please check the content of the not working file, I'm pretty sure it contains something unrelated.
Also, the fact that it is named `_tree` is suspicious.

Tz1rf, 2024-04-24:
@cecio 
For what its worth I also cannot replicate this problem. I can also confirm that there is nothing wrong with the firmware downloads. No matter how I download them they are the same. 

Anyhow @DFranzen 

Did you purchase the USB-C 4MB version? If so the following steps must be performed to load firmware on this unofficial Pico. As previously stated, the firmware should load and reboot. Probably useful for others that decide to go this route and purchase the unofficial Pico. 
  
1. Connect the Pico to your computer 
2. Long press the BOOT button 
3. Press the RST button
4. Release the BOOT button after 1 second;
5. If dont correct, the computer should show the storage on the Pico as a hard disk 
6. Copy the firmware file for USBValve to the pico and it should automatically reboot.

cecio, 2024-04-24:
thanks @Tz1rf

DFranzen, 2024-04-24:
Yes, I am sorry, I did not realize, that Right click -> save as, does not download the raw file, but an HTML-page. Maybe that's also what happened for Twanaki? Recompiling the firmware using the Arduino IDE, as they did, would also have solved my oversight.
Thanks for your help and time on this and generally for your work on this awesome project.

Can I suggest, in order to give this issue a constructive solution, to add a link to the Release page (https://github.com/cecio/USBvalve/releases) in the Readme-file in the instructions on "Flash Firmware".

Twanaki, 2024-05-14:
Hi guys,
You are right, I did the same mistake with right click and save as. Apologies.
Because the downloaded file is .uf2, I thought it was ok.
Sorry about this, have a nice day !

closed 2024-05-19 by cecio