cecio/USBvalve #32: Would Love PreBuilt One

jrock2004, 2024-04-10:
Did not see an issue or something but would love to see a pre-built one of maybe an amazon list of all the items needed. I think that would be cool

cecio, 2024-04-10:
thanks @NagelTuev

Tz1rf, 2024-04-18:
You can purchase the custom pcb, oled, usb port on eBay right now. It cost me 15.59 for it. Not sure what happen to the price of the Raspberry Pi Pico on Amazon. I bought a couple a few weeks ago and they only cost me 4.95 a piece, which is pretty good considering they cost $4 from the manufacturer when ordering them by the dozens.

jaharkema, 2024-05-04:
I would also love a prebuilt one. I just saw the LTT video and I think a device like this would come in really handy, but I am not at all comfortable with soldering (or PCBs for that matter).  
So just leaving a comment as +1 on the idea of a prebuilt USBvalve.

Tz1rf, 2024-05-04:
You can purchase a prebuilt unit on eBay.