cecio/USBvalve #34: Pico with built in display and usb-c

SeventhDread, 2024-04-10:
I’m quite interested in making my own USBValve and I’ve found this pico on Amazon https://amzn.eu/d/6Hgc5Mu would it work? 
It has usb-c which is what I was looking for and a built in bonus IPS screen. 

I also don’t like soldering I always make a mess of it so less is better so I would thinking I could get this but with the pin headers, I have cables that can connect to the pin headers then connect the correct pins to a usb a port and then try to 3D print an enclosure to hold it all. Is there anything I am missing or could this work?

Thanks!

SeventhDread, 2024-04-10:
Also building on this I was thinking of having usb an and usb c host ports would having both work with this? I saw in another post you said you could do it just one would work at a time, how would you wire this up?

And just being curious not related to what I’m asking what these detect cables like the hak5 O.M.G ones? And I’m guessing cables like them could be detected through either port (host or pico power port)??

cecio, 2024-04-11:
Hey @SeventhDread 

Thanks for your interest in the project. I try to address your questions below:
- the problem with all the Pi pico based boards out there is the LCD. You can try, but I'm pretty sure it will not work. Usually is not so hard to fix (you need to understand which is the LCD controller and find the correct wiring in terms of PIN),  but it will not work with  the provided firmware
- if you want to connect two different USB Host ports, you can do it, even if it is not really supported, especially because you are going to connect to the same PINs. You have to be careful because inserting two devices at the same time could potentially create damages, unless you put some diode/resistors on the connections
- I'm not sure about the Hak5 cables, never tried. If they act as HID, Mass or net devices, yes. If they are just hubs, may be not. If someone have one of these cables, I'm interested in knowing what happens when one is connected

Thanks,

SeventhDread, 2024-04-12:
Thanks for getting back to me! Seems the pico with a screen might just make things too complicated. Ill just go the normal route. Thanks, can't wait to make my own!

closed 2024-04-20 by cecio