cecio/USBvalve #35: [Enhancement] Write dump

Zorakie, 2024-04-13:
Hi,
This idea is probably one of the more complex ones:

It would be great to log possible read/write actions.

So basically:
1) Add a Card reader slot support (Quick search and it should be technically possible (https://www.youtube.com/watch?v=rr_JQchpa1Q)
2) On power up, mount the SD
3) If the device tries to write something through main (Micro/C) port, redirect it to the SD.
4) If the device on the USB-A port starts sending some data (HID), create a text file with random name and write the inputs.

That would help us to understand not only if the device is sending some data, but later on, we can remove the SD, put it into PC and find out, what has actually happened.

for MVP:
1) Fixed size card
2) Predefined format (FAT32).
3) On boot, check if card is inserted to enable logging
4) random name (as pPico does not have clock), only when creating files, prepend "host_" or "valve_" so we know through which end it came (and on PC, we then always can filter by file creation date)

Let's see, I have no idea how hard this would be even with borrowing some existing Pico SD project, and also I am not sure if the limited processing power of Pico would allow it, but anyway, it would be cool :)

cecio, 2024-04-13:
Thanks @Zorakie , it is a very interesting idea indeed.

And partially, it has been implemented in some way: if you connect the serial port created by USBvalve, you'll see that a partial dump of the disk operation is already printed out. Right now it is partial because the output on the serial is quite expensive and impacting timing of other operations.

But I think that it can be de-synched and probably managed completely: in this case if you log the serial output you may have the complete dump of the operations.

Adding and SD card is another option and it is feasible for sure, but it would require some redesign of the entire project. May be it could be  a mid-long term project.

Anyway, thanks for your suggestions, they are always interesting!!

Tz1rf, 2024-04-24:
Any reason this cannot just be logged to the nearly 1MB of free storage the Raspberry Pico already has? The storage that shows up when plugged into the computer and has a txt file already created in there that says nuke the entire site from orbit. Text files are extremely small so the 1MB storage should be plenty to lot a read out of read/write operations

Zorakie, 2024-04-24:
@Tz1rf
afaik, and cecio will correct me if I am wrong, that storage that you see normally is not a real storage, but rather a view of pico RAM. If you check the source code, those files are created in memory on startup, so once you unplug the pico, that memory block size to exist, which would include anything you would "store" there.

cecio, 2024-04-24:
what @Zorakie  says is absolutely correct.

@Tz1rf there are at least 2 reason for not using the internal USB storage:
- I need to monitor the activity on the drive at very low level, and I don't think it's possible without creating a "fake" layer with a storage on it
- in order to avoid to spread malware through USBvalve (which would be a shame ;-) ) everything is in RAM and destroyed at each unplug

Tz1rf, 2024-04-24:
@Zorakie 

This is why I suggested this storage space. I apologize if I described it incorrectly and caused a misunderstanding

@cecio 
I suggested the storage space for the reasons you pointed out. Its okay if the txt file is destroyed at each unplug. I wouldn't change this as its an appropriate air gap to prevent malware from spreading.  Or am I misunderstanding something here?

Zorakie, 2024-04-24:
Well, here is the problem: 1MB is enough for such scripts, sure. But the idea behind the dump is, you want to store the data on the go, then remove the card where you know you have a safe .txt dump, and review later.

Also the malware spreading is a valid thing.

And really, 90% of time, you will be somewhere outside, testing if some shopping mall charger is not trying to spoof your phone or something.
And even for USB-A testing, I would feel more safe running Pi from powerbank rather than testing on my PC (If I stumble upon killer USB, I will rather blow my 20€ pico project and 30€ powerbank rather than my 2300€ rig... 🤷‍♂️

cecio, 2024-04-25:
@Tz1rf sorry, I misunderstood your question. You are proposing to use the on board USB Mass storage device, just for logging purposes, correct?

If so: this area is accessible only in BOOTSEL mode from what I see, and it is not accessible in other ways. But I'll try to dig a bit more. The other accessible area is the flash, but it's small and not accessible as filesystem.

TryBreakFixAgain, 2024-05-14:
If added a SD-Card Logger on my Mods, Works like charm.  ;)

cecio, 2024-05-14:
oh wow, great job!!

Tz1rf, 2024-05-14:
@TryBreakFixAgain What is it able to log? 

I am very interested. If a custom PCB can be developed to use this SD Card add-on, I will make a new video showing how to build this. I wish I could help but I do not know much about creating gerber files for PCB, but based on the current one I would imagine adding the SD Card add-On to the back of the PCB with holes between the USB A Port and Screen might work.

TryBreakFixAgain, 2024-05-15:
@Tz1rf At the moment I write out the complete serial output to a static text file called datalog.txt, including a display warning if it is larger than 5mb at startup. I still have to improve the complete handling of the log files, but I am encountering some problems, so I have published this status for now.  A custom PCB should not be a problem, you can find all necessary infos in the readme of my fork in the point mappings. I will publish a log of an attack in the project wiki later today.

Tz1rf, 2024-05-15:
> @Tz1rf At the moment I write out the complete serial output to a static text file called datalog.txt, including a display warning if it is larger than 5mb at startup. I still have to improve the [...]

Thanks for the additional information. Is the problem you are encountering have to do with the limited processing power of the Raspberry Pico?

TryBreakFixAgain, 2024-05-15:
@Tz1rf  not only, some limitations are based on the SD lib its a hard combination for me, im only a beginner with microcontrollers and coding, and my free time is very limited