cecio/USBvalve #36: 3D Printable enclosure for 128x64 screen version

rtmq0227, 2024-04-17:
Since the only enclosure in the repository is for the smaller, x32px screen, I put together an option for the larger x64px screen (stl files attached).  It assembles with six M2x18 screws with room for locking nuts (links below), and some rubber padding to seat tightly without rattle.  It's designed for Resin printers, so you may need to file down and drill out the interior openings if you have a pronounced foot on the print.  I also recommend not soldering the screen to the PCB until you have it in the enclosure for the fit to be right.

STL Files: https://www.dropbox.com/scl/fo/kd9u046t1vtf3h2jkpmb5/ACpoGzTPq_6m6kwyjEtopGc?rlkey=sx72hl63xx694xxybgx68dki6&dl=0
Screws: https://tinyurl.com/mry88wyv
Nuts: https://tinyurl.com/yhzkmhmj
1mm Rubber Sheeting: https://tinyurl.com/4775dfs5

cecio, 2024-04-17:
oooh wow!! It looks awesome, thanks!!
I'll try to print it, even if I don't have a resin printer, let's see how it works.

rtmq0227, 2024-04-17:
I found the most success avoiding the elephant's foot issue ruining the face of it by orienting the main stl as shown (text facing up) but it does require supports on the underside in that case.  I originally designed it to print the opposite way but couldn't reign in the foot on my printer

rtmq0227, 2024-04-18:
I added a "simple" STL package for if the fiddly bits are too much for a filament printer.  I also added some better pics to showcase the final build better

Manofhunt, 2024-04-18:
I printed the original case on my FDM printer and the only issue I ran into is the holes for the nuts were a little tight. I was able to chamfer the holes with a razor blade and that solved the issue. Thank you for providing these! 
[image]
[image]

rtmq0227, 2024-04-18:
I'll admit I ran my tolerances a bit tight, I'm printing on a Photon Mono X.  I can adjust the "simple" version to add a little more room for the nuts, how much extra clearance did you need to add?

Manofhunt, 2024-04-18:
Difficult to say since I did it by hand, but probably about 0.5mm cumulative between the different sides I shaved on each nut location.

cecio, 2024-04-18:
@rtmq0227 thanks for your work on this.
If it's ok for you, I'd add the files to the repo. 
Thanks!

rtmq0227, 2024-04-18:
Gladly!

I'll update the simple files with the expanded nut holes tonight

rtmq0227, 2024-04-19:
Okay, I've added a v1.1 for the simple backplate with larger holes for the nuts, and here's an orientation guide (since Fusion always seems to export in the wrong orientation -__-)

[image]
[image]

cecio, 2024-04-19:
thanks @rtmq0227 
One question: am I wrong or in the new folder I see two version of the backplate but no spacer?

rtmq0227, 2024-04-19:
no, there should be all three. I removed the old version so now there's only three STL files in each version's folder

cecio, 2024-04-19:
ops, sorry. Thanks, I included your STLs in the STL folder and credited in the README.
Thanks again for your contribution.

pettard, 2024-04-25:
Really nice looking case! would it be an idea to also make a top plate for the 128x32 screen?

rtmq0227, 2024-05-01:
> Really nice looking case! would it be an idea to also make a top plate for the 128x32 screen? @rtmq0227

I wasn't going to, as there's already an enclosure STL for the 128x32, but a coworker was having trouble with that one so I may as well (better to have more options than fewer, right?).

It'll take time as I'll have to put together one with the smaller screen for reference purposes.

Tz1rf, 2024-05-13:
@rtmq0227 

Where did you make use of the rubber sheeting on the enclosure? Does it go between a layer?

rtmq0227, 2024-06-26:
@Tz1rf sorry for the delay in responding!  I use it between the middle layer and the PCB to take up the gap/tolerance and keep it from rattling around.  A small strip along each of the longer sides of the PCB do enough without messing with the ventilation.  I'll be doing an assembly video in the near future, but I'll try to get some pics for this thread in the meantime.

pettard, 2024-07-17:
> I wasn't going to, as there's already an enclosure STL for the 128x32, but a coworker was having trouble with that one so I may as well (better to have more options than fewer, right?).
>
> It'll take [...]

@rtmq0227  if you want me to test out the design i would be happy to do so.

rtmq0227, 2024-07-17:
@pettard So I actually did recently put one together, but the current STL has a second hole for a NeoPixel because I was trying to create a custom build with a color-coded indicator to match the events (green, yellow, red, flashing red).  I'll create a version without the extra hole and share it here soon.
[image]

rtmq0227, 2024-07-17:
@Tz1rf here's that image I promised:
[image]

pettard, 2024-08-12:
@rtmq0227 Any update on the 128x32 version?

rtmq0227, 2024-08-13:
> @rtmq0227 Any update on the 128x32 version?

Sorry, I lost a lot of personal bandwidth in preparation for DEFCON

Here's the STL of my x32 screen enclosure.  It works with my existing spacer and backplate STL files: https://www.dropbox.com/scl/fi/oqpq88l8r2xmb2uyov4sn/Enclosurex32-Final.stl?rlkey=n5luu3rnxmibmkgaoqbzm8br6&dl=0

Mikitwin, 2024-12-30:
For those who don't have a 3D printer, JLCPCB can make the USBvalve PCB as well as the enclosure (JLC3DP) at a very low cost. The only problem that I had is that, to avoid cracks or deformation, they ask for a minimum of 0.8mm wall thickness with the resin (Black resin) and the printing technology (SLA) I chose. The customer service is really nice so they warn you and ask for confirmation before they do the fabrication if this happens. For the 1.2_64 design for example, they sent me this thickness analysis:
[image]

To be safe, I switched to the 1.2_64_simple design but the screw holes were also too thin on that design. I accepted the risk and the result was still very clean!
[image]

With all that said, a possible improvement would be to update the designs so that there are no sections under 0.8 or 1 mm. This way people could order from JLC3DP without any issue. I'm afraid I'm not good enough in 3D design to make the changes myself. That's why I am very grateful that people provided designs for the enclosure!