cecio/USBvalve #38: Video Tutorial on how to build a USBValve

Tz1rf, 2024-04-18:
I thought I would post this here. I made a video tutorial on how to build a USBValve. Anyone should be able to follow this tutorial and build one at home. Feel free to post/share it and I hope those interested in this project find it helpful! [Youtube Video](https://youtu.be/7ymk8hD7-Hc)

cecio, 2024-04-18:
Wow @Tz1rf awesome work!
If it's ok for you I'd like to link it in the README file of the project.
Thanks a lot!

Tz1rf, 2024-04-18:
Yes you can link where ever you would like. You don't need my permission to distribute the video. Feel free to share it/post it anyplace you want.

LiPolymer, 2024-04-20:
@Tz1rf May I re-upload  your video on [BiliBili](https://www.bilibili.com/)?

YouTube has been banned in China,so that's pretty difficult to access the original link of your amazing video for Chinese users. :(

I'll mark the original link in the description. ;)

Thank you for your video :D

Tz1rf, 2024-04-20:
I would prefer that you don't. Re-uploading my video is not the same as sharing/posting it using the youtube link.

LiPolymer, 2024-04-21:
> I would prefer that you don't. Re-uploading my video is not the same as sharing/posting it using the youtube link. 

Thanks for your reply, I will not re-upload this video :D

Tz1rf, 2024-11-27:
I created a video showing how to flash the firmware to the Valve as well as how to use the device. https://youtu.be/Tp8xvrlqxUY

cecio, 2024-11-27:
Awesome! Thank you very much! I'm going to link also this one in the README. 
Thanks!