cecio/USBvalve #40: display not showing proper info

Drmalkav, 2024-04-24:
hi there, i tried to build this yesterday, but i can't get the display to show the proper text. I can only manage to get characters in the first "column" (pic1) i have tried to flash a couple of older versions too, but no difference. I'm pretty sure i've flashed it properly, because when i edit the readme file it gives me more characters on the screen (pic2). I'm still very new to the pico and i'd be gratefull for all help or tips you all have.
Sincerely, Henrik
ps- i haven't installed a female usb a to the bord, as i wasn't planning on testing bad usb drives yet.

IAmOrion, 2024-04-24:
I would suggest first running an I2C Scanner sketch to double check the I2C address of the OLED module.  The firmware uses 0x3C - it's possible your module uses a different address

Also double check the SDA/SCL connections are the correct and not the wrong way round.

Tz1rf, 2024-04-24:
@Drmalkav 
Can you please confirm wires are soldered on the correct pins?
PIN6 of Pi --> OLED SDA
PIN7 of Pi --> OLED SCL
PIN38 (GND) of Pi --> OLED GND
PIN36 (3V3OUT) of Pi --> OLED VCC

Drmalkav, 2024-04-24:
They are I triple double checked, even tried to reverse them as suggested.
I'm currently trying to figure out how to do the i2c scanner

Drmalkav, 2024-04-24:
> I would suggest first running an I2C Scanner sketch to double check the I2C address of the OLED module. The firmware uses 0x3C - it's possible your module uses a different address
> 
> Also double check the SDA/SCL connections are correct and not the wrong way round.

i got the scanner working on a Arduino nano, it says 
```
Scanning...
I2C device found at address 0x3C  !
done
```

cecio, 2024-04-24:
mmmmh....really strange. 
Could you try to re-solder the wires? May be a cold solder joint or a defective wire...it seems strange, but other than that, the only thing that remains is a defective screen. Did you tried it with other software?

IAmOrion, 2024-04-24:
> already tried resoldering, im waiting on some more screens in the mail, so i'll re test with those. Do you maybe know of some other software i can use to test my current display?

Try running the Adafruit SSD1306 Example "ssd1306_128x64_i2c" and see if that displays as expected.  If you don't have the library, you can install it via Arduino IDE or here: https://github.com/adafruit/Adafruit_SSD1306

You will need to change ```#define SCREEN_ADDRESS 0x3D``` to ```#define SCREEN_ADDRESS 0x3C``` on line 35

Tz1rf, 2024-04-24:
@Drmalkav 

Just for testing purposes did you try flashing the firmware for the smaller screen? It should still display text, it will just be larger and a bit pixelated.

Drmalkav, 2024-04-24:
@IAmOrion thx for that, i can now confirm, that my display works perfectly well :)
@Tz1rf  yup that looks like this 
[image]

Tz1rf, 2024-04-24:
Is this another unofficial Pico? Looks like a USB-C Port on there.

IAmOrion, 2024-04-24:
With USB-C that has got to be a clone, in which case the pre-built firmware may not work if the SDA/SCL pins are not broken out on the same GPIO/Pin numbers.

Drmalkav, 2024-04-24:
oh dang, i thought i had bought an original with just a type c port plopped on for convenience. Is there any way to figure out what the pin breakout is?
this is the description the seller posted
`Raspberry Pi Pico Board RP2040 Dual-Core 264KB Low-Power Microcomputers High-Performance Cortex-M0 Processor

Specification:

• RP2040 microcontroller chip designed by Raspberry Pi in the United Kingdom • Dual-core Arm Cortex M0 processor, flexible clock running up to 133 MHz • 264KB of SRAM, and 2MB of on-board Flash memory • Castellated module allows soldering direct to carrier boards • USB 1.1 with device and host support • Low-power sleep and dormant modes • Drag-and-drop programming using mass storage over USB • 26 × multi-function GPIO pins • 2 × SPI, 2 × I2C, 2 × UART, 3 × 12-bit ADC, 16 × controllable PWM channels • Accurate clock and timer on-chip • Temperature sensor • Accelerated floating-point libraries on-chip • 8 × Programmable I/O (PIO) state machines for custom peripheral support

Informaiton:

You can get more information for the Raspberry Pi Pico by this link or click the photo:`

IAmOrion, 2024-04-24:
> oh dang, i thought i had bought an original with just a type c port plopped on for convenience. Is there any way to figure out what the pin breakout is? this is the description the seller posted: [...]

Please provide a link to the product page from which you purchased this item

cecio, 2024-04-25:
[firmware.zip](https://github.com/cecio/USBvalve/files/15107551/firmware.zip)

@Drmalkav 
Could you try this version of the firmware? Unzip and load as usual. Thanks!

Drmalkav, 2024-04-25:
@IAmOrion this is where i bought it [aliexpress](https://www.aliexpress.com/item/1005006122200429.html?spm=a2g0o.order_list.order_list_main.84.e44d1802KcxBLG)

@cecio i got no image with that one.

Thank you everyone for your help and suggestions. I think i'll have to throw in the towel, and order a original board from a authorized seller.

tvandoorn, 2024-04-25:
Hello,

I don't mean to hijack this issue, but if @Drmalkav is giving up on this I wouldn't mind to keep trying. I am facing the exact same issue, but I am using the PCB, together with a Pico from [here](https://www.aliexpress.com/item/1005003371056277.html) (Color: Green), and a display from [here](https://www.aliexpress.com/item/1005005970901119.html) (Color: SSD1306 Y B).

[image]

I have tried the following so far:
* Running the 128x32 firmware
* Compiling the firmware myself using the Docker image
* Changing the I2C_ADDRESS to 0X78, 0X7B and 0X7A (based on what the back of the display says)
* The firmware made by @cecio

I can confirm the screen works by loading MicroPython on the Pico and running the following:
```
import machine
import ssd1306
from time import sleep

i2c = machine.SoftI2C(scl=machine.Pin(5), sda=machine.Pin(4), freq=400000)

print('Scan i2c bus...')
devices = i2c.scan()

if len(devices) == 0:
  print("No i2c device !")
else:
  print('i2c devices found:',len(devices))

  for device in devices:  
    print("Decimal address: ",device," | Hex address: ",hex(device))

oled_width = 128
oled_height = 64
oled = ssd1306.SSD1306_I2C(oled_width, oled_height, i2c)

oled.fill(0)
oled.text('Hello, World 1!', 0, 0)
oled.text('Hello, World 2!', 0, 10)
oled.text('Hello, World 3!', 0, 20)
oled.show()
```

[image]

The code above also prints the following I2C device information:
```
Scan i2c bus...
i2c devices found: 1
Decimal address:  60  | Hex address:  0x3c
```

cecio, 2024-04-25:
Could someone try the `HelloWorldWire.ino` sample from https://github.com/greiman/SSD1306Ascii ?
Thanks

IAmOrion, 2024-04-25:
For what it's worth, I have an official raspberry pico, the UF2 firmware also did NOT work for me - I didn't get anything on the display at all.

I used the Adafruit 128x64 I2C demo to make sure my OLED was working using 0x3C address and it was.

I then opened the source and uploaded directly to my Official Raspberry Pico using Earl Philhower's boards (https://github.com/earlephilhower/arduino-pico) and it worked without any changes.

So for whatever reason, the supplied compiled UF2 file found in firmware folder https://github.com/cecio/USBvalve/blob/main/firmware/USBvalve-0.17.0-64.uf2

did NOT work for me, but opening the .ino file found in https://github.com/cecio/USBvalve/tree/main/USBvalve (obv downloaded everything first, unzipped, then opened the ino from that folder) then uploading to my pico worked perfectly

Drmalkav, 2024-04-25:
@cecio here u go 
[image]

@IAmOrion just tried to follow your latest post, but got the same usual result

IAmOrion, 2024-04-25:
> @cecio here u go ![IMG_20240426_012117](https://private-user-images.githubusercontent.com/41089048/325796333-b02c54be-cf0b-4208-9dcf-280372d70304.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQwODc4NTYsIm5iZiI6MTcxNDA4NzU1NiwicGF0aCI6Ii80MTA4OTA0OC8zMjU3OTYzMzMtYjAyYzU0YmUtY2YwYi00MjA4LTlkY2YtMjgwMzcyZDcwMzA0LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI1VDIzMjU1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTcxNTUxMDk5YzIyNWVkNmU5Zjk2MWUzNGEzZWYwZGJmMjdiZjBkYjI1ZTNkMjQ3YmViMTFlNTNiYjIyMzllZjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.TiEEFBqm-RIEVdPFoswNsNBVBRxm_H24xocxV3Sxy-c)
> 
> @IAmOrion just tried to follow your latest post, but got the same usual result

You have the IDE & All libraries and board stuff installed? Have you got anydesk?

tvandoorn, 2024-04-26:
> Could someone try the `HelloWorldWire.ino` sample from https://github.com/greiman/SSD1306Ascii ? Thanks

I'm also getting just the exclamation mark on the screen with this example. When I replace `oled.print("Hello world!");` with the code below I only get the letter "o".
```
oled.print("H");
oled.print("e");
oled.print("l");
oled.print("l");
oled.print("o");
```

Looks like it only displays the last character of whatever is written to the display.

cecio, 2024-04-26:
Thanks to all for the tests.
From what I read it seems that the library used is not compatible with all the screen out there. I'll try to change something and see if I can improve compatibility.

cecio, 2024-04-27:
If someone have time do to a test:
[firmware.zip](https://github.com/cecio/USBvalve/files/15138135/firmware.zip)
It will probably not work, but I'd like to understand if the problem has something to do with the fonts.
Thanks

tvandoorn, 2024-04-27:
Hi, this is what it displays on the screen:
[image]

Drmalkav, 2024-04-27:
@cecio i get the exact same result with that firmware as @tvandoorn does.

@IAmOrion my soldering definitely needs improving xD but i can happily confirm, that there are no shorts :)

cecio, 2024-04-27:
thanks @tvandoorn @Drmalkav  , so it seems fonts are not the root cause :-)

cecio, 2024-04-27:
Tried to change something:
[firmware.zip](https://github.com/cecio/USBvalve/files/15139669/firmware.zip)
just check if display work, functionalities are not complete.
Thanks

tvandoorn, 2024-04-27:
Looks like you're onto something here 😁
[image]

Drmalkav, 2024-04-27:
Yup definitely did something 
[image]

cecio, 2024-04-28:
This should be a working release for 64 lines screen:
[firmware.zip](https://github.com/cecio/USBvalve/files/15141888/firmware.zip)
If someone can test it a bit it would be great. Next week I'll be a bit busy, so I will not make a new release, but if everything is fine, the week after I'll include this in the official branch.
Thanks

Drmalkav, 2024-04-28:
@cecio thank yoi so mich :) seems to work perfectly, it detects when I read the readme and when I write anything. I don't have a bad usb so can't test that part, but everything else looks awesome :)
[image]

tvandoorn, 2024-04-28:
Yep, it's also looking good here. I haven't received my USB port component yet so I cannot test bad USB, but I'll let you know once I do.

Thanks for your time and effort! Looking forward to seeing what caused this issue.

closed 2024-04-28 by Drmalkav

cecio, 2024-04-28:
Thanks to both @Drmalkav  @tvandoorn 
The issue was basically an incompatibility of the library I was using. I switched to a different one, but it was a bit longer because I had to implement some function not implemented in the new one (like scrolling).