cecio/USBvalve #41: [Question] File read attempt filename

IAmOrion, 2024-04-24:
Does this show the file name that was attempted to be accessed?

As an example, I have a device that auto updates from an attached FAT32 USB Drive assuming the correct file is placed at the root.  

If I was to plug this device into that USB connection, would the screen show, as an example, that "DEVICE_UPDATE.IMG" was attempted to be read?

Tz1rf, 2024-04-24:
@IAmOrion 

No currently the device cannot show that on the display. It would likely show !! mass storage, explanation points meaning the device is doing something that is unexpected. To see that information you would have to use the debugger on the serial port

IAmOrion, 2024-04-24:
> @IAmOrion
> 
> No currently the device cannot show that on the display. It would likely show !! mass storage, explanation points meaning the device is doing something that is unexpected. To see that information you would have to use the debugger on the serial port

ahh ok, but the debugger would show the file request then? (possibly)

Tz1rf, 2024-04-24:
My apologies, the display should also show that data was being transmitted, which is the indication that the USB Drive is trying to read or write data automatically. It just won't show specifics about this.  Yes the debugger possibly would shower the file request.

cecio, 2024-04-24:
Unfortunaltely this is not possible: the USB protocol just read and write data blocks, the files operations are done at Operating System level, so there is no way on the USBvalve to know the requested file: this is done by the operating system.

closed 2024-04-26 by cecio