cecio/USBvalve #42: fast charge detection

Xfaiv, 2024-04-28:
Is it possible to create a software modification to detect if the connected USB port supports fast charge? at least on a theoretical level, I'm not asking to activate the mode, I would expect that the Arduino would fry. I know of the existence of some USB testers that are capable of this.

Thanks in advance for your trouble

Zorakie, 2024-04-29:
Would that be even possible? I know from some older articles 'maybe things have changed...?), that basically every manufacturer implements their own proprietary protocol, so even if on paper the adapters may be the same in terms of specs, afaik, if you take like Xiaomi charger and you try to charge Samsung phone, it will charge, but slowly, not activating the fast charge, just because the phone can not agree with that vendor-locked charger on the voltage, amperage, etc...

cecio, 2024-04-30:
Hey, thanks for your question. TBH I’m not super aligned on the matter and this looks like more a feature detection question, than a security one (and USBvalve is meant to be a security device). 
So let’s say is not on the roadmap, but I’ll keep this in mind anyway. Thanks

closed 2024-05-03 by cecio