cecio/USBvalve #44: Button Press Required on Boot?

Rae-Hughes, 2024-05-05:
Hi There, I absolutely love the USBvalve! I have mine mostly working but each time I power it up I have to press the "reset" button before the screen will turn-on. Is this part of the design, and if so is there a workaround? Or do you think I've missed a solder somwhere?

Thank you so much for developing this project :o)

NagelTuev, 2024-05-05:
it shouldn't really be like this, re-solder all the soldering points and flash the valve again

cecio, 2024-05-05:
Hey @Rae-Hughes thank you very much for the kind words!

No, this is not expected. It should start without pressing reset. What @NagelTuev (thanks!) says is correct.
If it does not help, please provide some more info:
- firmware version
- LCD version (32 or 64)
- Pico Board used (official model, clone, other....)
- LCD hardware used
- did you wired your connections or used the PCB?

Thanks,

Rae-Hughes, 2024-05-05:
Hi Folks, thank you so much for getting back to me so quickly! I'll re-solder and get back to you. My soldering skills aren't great so I thought that might be the issue

Rae-Hughes, 2024-05-08:
Hi hi,

So I started again, fresh PCB, New Pico, new OLED, new USB port. Soldered everything. Flashed the Pico. Yet I still have exactly the same issue. I need to press the button to get the screen on when I first power it up.

Any thoughts?

cecio, 2024-05-08:
Which firmware version are you using? The latest? Could you try to flash an older one, like 0.17.0 and/or 0.16.0?
Also, is your Pico a standard one or a clone? Some of the clones has additional features that can introduce incompatibilities.
Thanks,

Rae-Hughes, 2024-05-09:
I tried flashing with both 0.17.0 and 0.16.0 and they both had the same issue. My Pico is a standard one as far as I'm aware

cecio, 2024-05-09:
mmmmhhhh...weird...
Let's try to recap:
- you insert the cable in the Pico micro USB
- then you connect the cable to the PC
- you don't see anything on the screen nor a device is mapped on the PC
- then you press reset
- at this point the screen turns on and you see a mapped device on the PC

Could you confirm that this is what is happening?

Rae-Hughes, 2024-05-10:
So what happens is,

- I insert the cable into the Pico micro USB
- I connect the cable to my PC
- I see a mapped device on the PC as normal (with the .txt file, etcetera)
- The screen shows nothing
- I press the reset button
- I see a mapped device on the PC *and* the screen turns on and then functions as normal

cecio, 2024-05-10:
which is the model of the screen you are using? Do you have a link ?

Rae-Hughes, 2024-05-11:
Hi, I'm using this,

https://www.diymore.cc/products/diymore-0-91-inch-iic-i2c-oled-lcd-12832-128x32-display-diy-module-ssd1306-driver-ic-dc-3-3v-5v-stm32-for-arduino-pic?variant=14646817521722

cecio, 2024-05-12:
Can you try the following firmware and let me know if it works?

[firmware.zip](https://github.com/cecio/USBvalve/files/15285714/firmware.zip)

Thanks,

Rae-Hughes, 2024-05-15:
Hi hi,

The firmware does function, but it doesn't solve the issue with the button..

TryBreakFixAgain, 2024-05-15:
i ran into the same problem and realised that it was sometimes due to bad connections, but also partly due to the power supply, especially with the very cheap displays.

how do you connect the display to the pico? by cable, pcb or breadboard? Maybe your cables are not in good condition, or the pcb. you could use a multimeter to check the continuity and the resistance of the connections, with the short cables there should be almost no resistance.

Rae-Hughes, 2024-05-16:
I'm using a PCB, so maybe it's that? But it's a well-built PCB..

I might have to invest in a better OLED

TryBreakFixAgain, 2024-05-16:
Do you know how to use the arduino ide? install the adafruit ssd1306 library, in the examples you will find the ssd1306_128_32_i2c.ino install it on your pico. if it gives the same error, it is the hardware.

if you can't get it to work, let me know and I'll build you the firmware quickly.

alternatively you can also try to solder the display with cables to test if it's the pcb. But I think it's more likely to be the displays.

Hephalumpo, 2024-05-16:
I tried, and was unsuccessful. If you've got a version of the firmware I could download that would be super helpful

TryBreakFixAgain, 2024-05-16:
[ssd1306_128x32_i2c.ino.zip](https://github.com/cecio/USBvalve/files/15338862/ssd1306_128x32_i2c.ino.zip)

no problem ;)

Hephalumpo, 2024-05-16:
Thank you so much!! I have the same issue, the screen doesn't light when I plug the pico in

TryBreakFixAgain, 2024-05-16:
Sounds like hardware issue, do you get something somehow, or nothing at all? if u get nothing your displays could use an other I2C adrress, but i dont think so

closed 2024-07-06 by cecio