cecio/USBvalve #48: printf errors when compiling in Arduino IDE 2.3.2

rtmq0227, 2024-07-16:
I'm trying to add some features to the current build (ARGB/NeoPixel support) and after configuring the IDE with the libraries and versions you listed, I keep getting an error on the serialtinyusb.printf() calls, suggesting I use print instead.  Is there another version of tinyusb you're using or any custom source files you may have injected?  I already had to add the stdlib.h and stdio.h files to Pico Pio manually as they are no longer included in the package the IDE downloads.

cecio, 2024-07-16:
hey.

No, I don't recall any other library added to the environment. It's strange that you had to add that header files, this is a sign that something is wrong, may be some conflicts between libraries.
To install Pico PIO USB you should add it to Library Manager (https://www.arduino.cc/reference/en/libraries/pico-pio-usb/). Remember to use version `0.5.3` and not the latest, which is causing some issues.

To simplify things, you can use the docker file to compile, it should download and build the correct environment, you can follow instructions in the README for that.

Cheers,

cecio, 2024-07-17:
I tried to make a new installation of the IDE from scratch.
The only thing  I had to manually modify was to add 

https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json

in the `File-->Preferences-->Additional Board Manager URLs` in order to be able to install the proper board (`Raspberry Pi RP2040 (3.9.3)`). Then I just added the libraries with the library manager (installing with dependencies) and it worked, without modifying anything else.

Let me know if this helps.
Cheers,

rtmq0227, 2024-07-18:
That absolutely did the trick!  Thank you so much for checking back in on this.

Now I need to figure out why the Adafruit_NeoPixel code I added is crashing the device, but that's an issue for another time.... :)

closed 2024-07-18 by cecio