cecio/USBvalve #49: Screen not working (black dots and white lines)

markthwain1797, 2024-07-16:
Me again,
this is my first breadboard build an I never really used microcontrollers before, so please don't be harsh.

I got [this screen](https://www.az-delivery.de/products/1-3zoll-i2c-oled-display) in addition to a rasperry pi pico.

My screen looks like the following is no device is connected vs if a bad usb is connected:
[image]
[image]

I'm not exactly sure I'm on the right path, but my research showed it might be a problem with getting the right driver for "cheap" OLEDs: [Link (reddit)](https://www.reddit.com/r/esp32/comments/10v4dmr/comment/j7fegwj/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button)

Any help or suggesting is very appreachiated. Thanks in advance.

cecio, 2024-07-16:
Hey.

Which firmware version did you uploaded? With this screen you should use tge 64 lines version:

`USBvalve-0.18.2-64.uf2`

or if you are compiling from code, you should change line 86 from

```
#define OLED_HEIGHT 32
```
to 
```
#define OLED_HEIGHT 64
```

markthwain1797, 2024-07-17:
Hi, thank you :)

I actually tested both versions. The pictures above are from the 32 lines version.

If I use the 64 lines version the screen looks like this (without and with bad usb)
[image]
[image]

cecio, 2024-07-17:
Yes, it looks like a compatibility issue with the OLED.
Can you try to use one old firmware and in particular https://github.com/cecio/USBvalve/releases/download/v0.17.0/USBvalve-0.17.0-64.uf2

This version was using a different OLED library, give it a try and let me know.

cecio, 2024-07-17:
ok, I probably found the reason: your display use the `SH1106` chip. The library currently used by USBvalve is the `Adafruit SSD1306` and it seems it does not support the other chip:

https://github.com/adafruit/Adafruit_SSD1306/issues/14

Try to use the old firmware as suggested in my previous message, it may work. If not, you probably need to switch the library. I don't know if you are familiar with development or not.

I may consider to switch the library to something with wider support, but  I'd like to find something with a solid installed base.

markthwain1797, 2024-07-18:
The version you suggested (v0.17.0) seems to work perfectly fine for me.
I'd be really interested in testing if you try out different librarys that support types of displays.

cecio, 2024-07-18:
OK, great! I'm happy it worked.

I switched to this new library to improve compatibility, but it seems in some cases the old one was better.

I'll try to have a look if I can keep compatibility for both.