cecio/USBvalve #50: Missing parts for USB Pipe

TheLumpOfDirt, 2024-08-03:
I was looking at the BOM for USB Pipe and I noticed that you are missing a listing for resistors R7 and R8.

cecio, 2024-08-03:
Thanks for the notification, I forward this to @LiPolymer for checking.
Thanks

LiPolymer, 2024-08-04:
> I was looking at the BOM for USB Pipe and I noticed that you are missing a listing for resistors R7 and R8.

I'm sorry for that.
R7 and R8 aren't actually connected to anything. They are added for circuit debugging purposes
[image]
[image]
I'll note this in the README and BOM.
Apologies again :(

cecio, 2024-08-04:
Fixed with Pull Request #51 
Thanks a lot @LiPolymer  for your fast feedback!

closed 2024-08-04 by cecio

LiPolymer, 2024-08-04:
That's my duty ;)