cecio/USBvalve #52: UF2 file not booting

Onexkings, 2024-10-15:
Hi everyone,

I’m still new to this USB valve project, and i have encounter an issue. I’ve flashed a UF2 blink test file to my Raspberry Pi Pico to make sure it was working, and it works fine. However, when I try the USBvalve-0.19.0-64.uf2 file to the Pico, nothing happens—the OLED display remains off, and I’m not seeing any output.

Here’s what I’ve checked:

The Pico is in bootloader mode when flashing the UF2 file.
I resolder pin GND2 and VCC2

Since im new to this, i dont know if i need a libary or another aplication to work with this Pico
If anyone has experienced similar issues or has suggestions on how to troubleshoot this further, I would really appreciate your guidance!

Thank you!

cecio, 2024-10-15:
Hey.

No, you don't need anything else to use it.
I would check if it's a display issue: once you flashed the USBvalve firmware, could you try to attach it to a PC? Do you see any attached device if you do so? For example in Windows you should see a new USB device attached, in Linux you may have to mount it manually, but you should see a new USB device as well.

In this way we can understand if the problem is the display compatibility/connection.
Let me know,
Thanks,

Onexkings, 2024-10-16:
Hello,

Thank you for replying, I've bought a new pico as well an oled to makesure that its not the pico problem aswell as an oled issue. Before i asemble it to the PCB, I've tested it out using a breadboard and it works! im worried that if i assemble it to a PCB, it won't work again, but im not sure and haven't tested it yet. 

But, I really appreciate your guidance and support as I navigated this project as a beginner. Thank you once again for your help!

Thank you!

Onexkings, 2024-10-16:
Hello,

I've solderd everything to the PCB and same issue the oled wont display, so could it be the PCB problem here?

cecio, 2024-10-16:
It could be different things, but first of all we need to understand if the USBvalve firmware is working (except for the OLED). So, after you upload the firmware, do you see a new USB device on your system?
If so, the problem is in the OLED, and it could be:

- incompatible OLED. The OLED should be based on `SSD1306`. Other may work, but not always.
- OLED connection not correct. Check again if you soldered the right PINs

In case of incompatible display, you can have a look to issue #49 and may be try the firmware version 0.17.0

Let me know how it goes.

Onexkings, 2024-10-26:
Hi, 

I apologize for my delayed response, I nearly forgot about my question. I also appreciate your patience as I provide an update regarding my issue.

I’d like to answer your question now. Yes, I see a new USB device on my system.

I found the issue after soldering everything to the PCB. The new Pico and OLED worked fine on the breadboard, but the OLED didn’t turn on when soldered to the PCB. I connected a jumper wire from the Pico pin directly to the OLED pins, and that fixed it. I had the same problem with the USB not providing voltage, so I also jumped a cable from the Pico to the USB socket, and it worked!

If I'm not mistaken, the issue seems to be with the PCB wiring diagram. Or it could be my pcb manufacture problem?? but im not sure. I did put a blob on the OLED power pads, but there was still no result.

Thank you for your assistance!

cecio, 2024-10-26:
no worries, thanks for the additional details.

The PCB wiring diagram is fine, I built hundreds of them with the same diagram. It could be a problem in the manufacturing  of your PCB, but it's unlikely. 

Try to have a look at this [video](https://www.youtube.com/watch?v=7ymk8hD7-Hc). At minute 8:09, it explains how to solder the power pads:
- be sure that you put solders on only two of them
- be sure that they are correctly soldered (sometimes these little pads seems soldered, but if you look carefully with a magnifier, they aren't)
- be sure to solder the right two (which, in most cases, are the same two shown in the video. You can understand which are the correct two by checking the README)

Let me know if this helps.

cecio, 2024-11-19:
stale

closed 2024-11-19 by cecio