cecio/USBvalve #53: Feature request: Counter for input strokes

dxniro, 2025-01-13:
Hi!

First of all: Thank you for the cool project you set up! :)

What do you think about having a counter on the display for HID devices, which shows how many inputs were made (for example keystrokes). This would be helpful especially if you want to have an unknown usb device connected for a longer period of time to check whether it is doing something anomalous after a specific time or not.

The counter could probably be placed next to the "HID Device sending data... (counter)" and could be refreshed every one minute (or every time, an new input comes in). I was thinking of two variations: 

- Refresh the display & keep the counter in the same line
- Append a new line after an specific period of time with the refreshed counter

What do you think about this? Would this maybe be a useful feature?

Best regards,
Daniel

cecio, 2025-01-13:
Hey Daniel! :-)

Thanks a lot for your kind words and for participating actively to the project with your request!

Your proposal is interesting, but let me try to better understand the need.

In my mind I have two main use cases when I think to the USB devices detection:

- detached usage (so, no PC available, just the *USBvalve* device): in this case I consider to attach the unknown device to *USBvalve* and let it attached to it for the time I need (one second or one hour). In this case if I see any kind of activity as HID device (no matter if 1 or 1000 char), I would consider the device suspicious and I would not attach it to anything important
- attached usage (PC available): in this second case I can attach to the serial device from the PC and monitor the complete activity of the HID device also displaying any output coming from it (like [this](https://asciinema.org/a/NWfC9Mvzzpj3eZfsC7s5Dz1sJ)). So even in this case, knowing the number of keystrokes on the panel, is not adding a critical information

But that said, I'd like to know why you consider this information useful and in which case it would be worth to have it available on the OLED screen. May be there is something I didn't considered. :-)

Thanks a lot for your time and for your interest in the project!

dxniro, 2025-01-15:
Hi cecio,

thank you for your feedback!

The case is maybe really rare, but there would be a small change, that you could plugin an device, which you expect to be an HID device, and that it will do nothing suspicious, until you started to use it. Let's say you have an untrustworthy keyboard, which acts like a rubber ducky, but with the difference, that it is waiting, until you started to use the keyboard or to type the first few key strokes.

If USBvalve would show me, how much inputs were already made I could type in something randomly and then let it plugged in and not touch it for let's say an hour. When the counter increases, I know that this is really suspicious. With the current state, it would always show me only "sending data".

Also if you plug in a mouse, which is really sensitive, you could easily trigger the "HID device sending data..." by just plugging in the mouse. With the counter I could keep track of it, whether there are more inputs coming in or if it was just me moving the mouse by accident.

Hopefully, this explains it better to you, what I thought about it, when I wrote this request :)

Have a nice day!

cecio, 2025-01-15:
Thanks for your additional thoughts!

Ok, I got your points. I'll give it a try to see how it may work.

Thanks for your input so far.

Have a nice day you too!

cecio, 2025-01-28:
I tried different solution for this, but this is going to clutter the tiny OLED.

So, I'm thinking to make this optional and printable with a long press of the BOOTSEL button. So at the end it will be:
- short press (under 2 seconds): RESET (as it is right now)
- long press: print HID events number

dxniro, 2025-01-30:
That sounds good to me! :)

cecio, 2025-02-02:
Implemented in release `0.21.0` (d756e20eb4a4a87d770daf3bcb9ca3a2fa89ac30)

closed 2025-02-02 by cecio