cecio/USBvalve #54: No screen output despite successful flashing

FabianKnebel, 2025-02-08:
Hi,

First of all: great project, thank you!

I'm fairly new to DIY projects and think I need some help with my problem:
I can't get any output on the OLED screen.

It seems like the firmware is working correctly because:
I successfully uploaded the firmware and the custom drive appears.
Also it seems to run, because the LED on the Pico is on and turns off when I connect my keyboard to the USB host.
However, the screen stays black all the time.

What I’ve Tried:
- Tested the wiring and screen (the x64 one) with a separate script, where the screen displayed the test correctly. So this shouldn't be the problem.
- Tried both 128x64 and 128x32 OLED screens with their respective firmwares. The issue occures on both screen sizes.
- Tested older firmwares (down to 0.17.0), but the issue persists.

I’m attaching a photo of my wiring for reference.
[image]

Any idea what I might be doing wrong?
Thanks in advance for your help!

cecio, 2025-02-08:
Hey!
Thanks for your interest into the project!

From what you say the issue seems to be a compatibility with your OLED. Could you check if your screen is based on `SSD1306` IC? If not, this could be the reason.

FabianKnebel, 2025-02-09:
Thanks for the quick reply.

This is the [display](https://www.az-delivery.de/products/0-96zolldisplay?variant=26462805705) I use.
At least it has `SSD1306` in the product name.. Also I used the SSD1306 library in my test to check the wiring. And this worked.

cecio, 2025-02-09:
OK, try to play a bit with this value (line 83):

```
#define I2C_ADDRESS 0x3C 
```

Looking at the link you sent me, it looks it uses something like `0x78` or `0x7a`. Then you need to recompile. I suggest you to use the dockerfile included for that, it will simplify a lot the process. Or, if you struggle in doing that let me know, I can provide a couple of custom firmware.

FabianKnebel, 2025-02-09:
Thanks for taking a closer look.

I briefly tried to recompile it, but failed the process. And it's not something I can dig deeper into now.

It would be great if you could provide custom firmware!

cecio, 2025-02-09:
I'm attaching two versions for the 32 lines. Let me know if one of them work.

[fw.zip](https://github.com/user-attachments/files/18724130/fw.zip)

cecio, 2025-02-09:
I realize now that you showed the link for the 64 lines version, so it's probably better to test this one :-)
Here the 64 line versions:

[fw_64.zip](https://github.com/user-attachments/files/18724170/fw_64.zip)

FabianKnebel, 2025-02-09:
Thank you for your effort, but I got the same behavior. The on board LED acts like expected, but still no output on the OLEDs for me.

I tried both versions of the 32 line and 64 line displays - even with another Pico and display.
Also I can proof, that in principle my wiring and the screen itself work:
[image]

Any other ideas? Am I maybe using wrong boards ([these](https://www.amazon.de/dp/B09TSWP4RV))?

cecio, 2025-02-09:
I see.

Which code are you using for the "Hello World" sample? One of the standard examples? Can you identify the I2C address used to initialize the screen?

cecio, 2025-02-09:
Try this one (64 lines), may be we are lucky.

[fw_3d_64.zip](https://github.com/user-attachments/files/18724597/fw_3d_64.zip)

FabianKnebel, 2025-02-09:
This sadly also didn't help.

This is my testing code:
```
from machine import Pin, I2C
import ssd1306

# Initialize I2C
i2c = I2C(1, scl=Pin(7), sda=Pin(6))

# Create the display object (width=128, height=64)
oled = ssd1306.SSD1306_I2C(128, 64, i2c)

# Clear the screen
oled.fill(0)
oled.show()

# Display text
oled.text("Hello, World!", 0, 0)
oled.show()
```

Is this correct to identify the I2C address?
```
from machine import Pin, I2C

i2c = I2C(1, scl=Pin(7), sda=Pin(6))
devices = i2c.scan()

if devices:
    print("I2C devices found:", [hex(device) for device in devices])
else:
    print("No I2C devices found.")
```
Output:
`I2C devices found: ['0x3c']`

cecio, 2025-02-09:
Looking closer at the picture you posted, it looks you are using the wrong PINs for SDA and SCL.
If I count the PIN you are using 9 and 10, but you should use 6 and 7.
Try to double check this

FabianKnebel, 2025-02-09:
OMG, you are absolutely right!
I thought it was GP6 and GP7, that's why I used PIN 9 and 10..
Now it works with the default firmware.

Thank you anyway for the support!

closed 2025-02-09 by FabianKnebel

cecio, 2025-02-09:
Great! Happy it is solved now!