cecio/USBvalve #55: Pico 2 crashes

deriggsm, 2025-02-19:
HID device and microboard issue

deriggsm, 2025-02-25:
Pico 2w, causes disconnected causes a crashes works properly when connected. Could be Wifi or blue tooth

https://github.com/user-attachments/assets/cf6cab32-0a72-407b-a57a-715c72d4eb21

cecio, 2025-02-25:
mmmhhh, I see.
The output you see (the sequence of  `(0 0 0)`) should not happen, because in the code every print of this string should be preceded by a `MOUSE:`  string. That means that something broken the code flow.

Did you recompiled the code before uploading it to the `Pico 2W`? Because the provided firmware is compiled for the standard `Pico 2` and the two formats are  different.
So my best guess right now is that, if you uploaded the firmware provided in the repository, it is not compatible. You should recompile it for the `2W`.

Let me know if this is the case.
Thanks

closed 2025-03-22 by cecio