cecio/USBvalve #56: Nothing happens when plugging USB drive to USBvalve

Mikitwin, 2025-02-27:
I built my USBvalve with the PCB thanks to Tzirf's tutorial. However, when I plug a USB drive, nothing changes on the USBvalve screen. I only get "USBvalve - 0.19.1" and "Selftest: OK". I would expect a "Mass storage" message when looking at Tzirf's second tutorial. There is also no light turning on on the inserted USB drive. I checked a few PCB connections with a multimeter already but didn't find anything obvious for now. The firmware seems to works since I can modify the text file if I plug the USBvalve to my PC and I get a warning on screen. I'm suspecting a problem with the USB connector, but I'm not sure what to look for. Any help would be much appreciated!

Mikitwin, 2025-03-01:
Hi,

I tested a mouse and a keyboard, but I had the same issue. I checked the traces and soldering more carefully with a multimeter to realize that I was a bit light on solder for the USB connector and the signals were not going from the PCB to the USB connector. Rookie mistake... I added more solder and flux but same issue. Then I just tried to press the reset button and it worked! My USB drive was not recognized automatically when I inserted it into a powered USBvalve, but it was if I pressed the reset button. I only realized later that it was a boot drive, so I don't know how it interacts with the USBvalve. Using another regular USB drive fixed the problem. Thank you for your help and for creating this amazing project!

cecio, 2025-03-01:
Thanks to you for your feedback!

closed 2025-03-01 by cecio